weisongwen/UrbanLoco #1: Downloading problem

TakahashiJinxu, 2020-07-13:
Over 48 hours ago I attempted to download a sequence of the UrbanLoco dataset. I filled in and submitted the form as required by the website, but didn't receive the link within 48 hours.  Is it because my email address didn't pass your checking? I agree with all the terms of the dataset, including using it only for non-commercial purposes. Could you please check whether you could send a downloading link to me? Thank you!

yzhou377, 2020-07-13:
HI Takahashi-san, I have send you an email on Friday regarding you download request. It seems you had the wrong link. The new link is copied with the email. Thank you for your interest in our dataset!

weisongwen, 2020-07-13:
Hi @TakahashiJinxu,  Meanwhile, the google form is updated. You will see the download link after you fill the google form.

Thank you for your interest in our dataset!

TakahashiJinxu, 2020-07-13:
> HI Takahashi-san, I have sent you an email on Friday regarding you download request. It seems you had the wrong link. The new link is copied with the email. Thank you for your interest in our dataset!

Thank you very much! I have found the download link. And the link was sent to my gmail instead of the mail address I submitted via the form.

closed 2020-07-13 by weisongwen