weisongwen/UrbanLoco #3: Why there is no camera data in California dataset?

YiChenCityU, 2020-07-17:
[image]

weisongwen, 2020-08-24:
@YiChenCityU  Since there is no response after our email contact. We mark this issue as fixed. Thanks for your interests in our datasets.

closed 2020-08-24 by weisongwen

TakahashiJinxu, 2020-09-01:
Hello! I also found that there is no camera data timestamp in California version. Namely, when I tried to extract the camera and IMU data form the ROS bag using Kalibr (after a little modification to support compressed_imgmsg), I found the timestamps of all images are zero. The command I used was
kalibr_bagextractor --image-topics /camera_array/cam0/image_raw/compressed --imu-topics /imu_raw --output-folder dataset --bag ../CA-20190828194706_blur_align.bag
I also have tried UrbanNav-HK-Data20190428 in urbannav dataset and successfully got the correct timestamps when extracting the data by Kalibr. How can I get the timestamps of the images in California version in this dataset? Thank you in advance!