weisongwen/UrbanLoco #5: How to get the timestamps of the images in California version in this dataset?

TakahashiJinxu, 2020-09-02:
Hello! I also found that there is no camera data timestamp in California version. In fact, I noticed the existing but closed issue #3 which is almost about the same question, but didn't  find out the solution.
It goes as this. When I tried to extract the camera and IMU data form the ROS bag using Kalibr (after a little modification to support compressed_imgmsg), I found the timestamps of all images are zero. The command I used was
kalibr_bagextractor --image-topics /camera_array/cam0/image_raw/compressed --imu-topics /imu_raw --output-folder dataset --bag ../CA-20190828194706_blur_align.bag
I also have tried UrbanNav-HK-Data20190428 in urbannav dataset and successfully got the correct timestamps when extracting the data by Kalibr. How can I get the timestamps of the images in California version in this dataset? Thank you in advance!

weisongwen, 2020-09-04:
@yzhou377  Yiyang, could please kindly help to fix the issue?

rwn17, 2020-09-05:
Come across the same issue. There is no data in the header of img_msg.

yzhou377, 2020-09-10:
I concur with your observations of missing timestamps in headers. The missing timestamps came with a reason: in order to re-process the image data to protect residential privacy, all camera messages are post-processed with a segmentation algorithm to blur the human faces and vehicle plates. When republishing these topics, we did not publish the re-processed timestamp with it as it was a few weeks after the real recorded time. 

However, the camera and the LIDAR are well synchronized with a hardware trigger plus post-processing. In this BLURRED rosbag you are having now, the two sensors are synchronized already. If you would like to see the actual timestamp of them, I would suggest you use the recorded timestamp, which is inherent to the ROSBAG, not a specific message. Here is a quick tutorial for you: https://answers.ros.org/question/82091/how-to-associate-rosbag-time-with-data/ This should be sufficient if you indeed want to match one LIDAR message with one camera message. 

I have personally used this method during our post-processing for privacy concerns, so I think it would be a good start for you as well.

TakahashiJinxu, 2020-09-11:
> I concur with your observations of missing timestamps in headers. The missing timestamps came with a reason: in order to re-process the image data to protect residential privacy, all camera messages [...]

Thank you! Using the recorded timestamp inherent to the ROSBAG as you suggested really solves my issue.

weisongwen, 2020-09-11:
@yzhou377 Thanks Yiyang. 

@TakahashiJinxu @rwn17 Solved and Closed. Please re-open the issue if you got issues.

closed 2020-09-11 by weisongwen