weisongwen/UrbanLoco #6: Downloading problem: Do HK-Data20190117 and HK-Data20190317 need to be shared using sharing link?

TakahashiJinxu, 2020-09-03:
Hello! I have been trying to download the data, but failing every time I try to download HK-Data20190117 and HK-20190317. I can successfully start downloading other sequences. I examined the download links, and discovered that the links of  HK-Data20190117 and HK-20190317 do not end with "sharing". Are they private links so that I can't download? (I understand it may also due to my network problems.) Thank you!

weisongwen, 2020-09-04:
@TakahashiJinxu  I think it is your network issues since I can download it in several computers from my end.

TakahashiJinxu, 2020-09-04:
> @TakahashiJinxu I think it is your network issues since I can download it in several computers from my end.

Thank you! It may indeed due to my network issues.

weisongwen, 2020-09-04:
@TakahashiJinxu . Thanks for your update

Closed and Fixed :-)

closed 2020-09-04 by weisongwen