weisongwen/UrbanLoco #7: How to get the 6DoF Ground truth poses at the frequency of LiDAR?

robotstar, 2020-10-17:
This is a very valuable dataset for SLAM researchers due to that it provides Ground truth poses for quantitative evaluation.

In the rosbag file, the topic name of Ground truth pose is probably '/navsat/odom', whose frequency is far lower than that of 'velodyne_points'. 

For convenient evaluation, the Ground truth poses at the frequency of LiDAR are required (e.g. KITTI dataset). How can I get the 6DoF Ground truth poses at the frequency of LiDAR?

Thank you!

weisongwen, 2020-11-04:
Hi @robotstar ,

Thanks for your interest and sorry for the late reply. For the HK dataset, the frequency is 1Hz which is smaller than that of the Velodyne sensor. Therefore, you can only evaluate the performance of the 3D LiDAR SLAM with a frequency of 1 Hz. The good thing is that the CA dataset provides a higher frequency of ground truth. Meanwhile, the ground truth is provided by the topics ```/novatel_data/inspvax```. 

Hope this helps!

Best,
Weoisong,

weisongwen, 2020-11-12:
Closed as there is not reply for a long period. Feel free to reopen the issue if you get any issues.

Thanks,
Weisong,

closed 2020-11-12 by weisongwen