weisongwen/UrbanLoco #8: VINS MONO config file

Sean0120, 2020-10-27:
Hi,
    I want to run VINS-MONO on the data you provided. But I find that there are some parameters loss such as imu paramters and td. Also even if I fill in them manually, VINS-MONO would reboot from time to time. Can you provide the config file you used for VINS-MONO?
    Thanks for you attention and help.

weisongwen, 2020-11-04:
Hi @Sean0120  which dataset are you using? The numerous dynamic objects in our dataset can be a large challenge for the VINS-Mono. You can try some easier datasets with a few dynamic objects. 

To run the VINS-Mono demo, I suggest using the CA dataset and setup the noise parameters of IMU directly based on default parameters in VINS-Mono. My experience is that IMU the parameters in VINS-Mono is highly related to the parameters for the image reprojection.

Thanks,
Weisong,

weisongwen, 2020-11-12:
Closed as there is not reply for a long period. Feel free to reopen the issue if you get any issues.

Thanks,
Weisong,

closed 2020-11-12 by weisongwen