weisongwen/UrbanLoco #10: question about 田河

huanhexiao, 2020-12-15:


weisongwen, 2020-12-16:
@huanhexiao  thanks for the interest. " [to LIDAR]" means sensor_T_LiDAR. In other words, all the sensor measurements are calibrated to the LiDAR sensor on the roof of the vehicle. 

Regarding the 7.62e-02, this means that the z-axis is almost the same with respect to the LiDAR and IMU. I am very sure that our calibration accuracy extrinsic is less than 7.62e-02. 

Hope this helps. BTW, the recent papers using our datasets seem to be ok with the provided extrinsic. You can also take a look at those papers.

Hope this helps,

huanhexiao, 2020-12-16:
> @huanhexiao thanks for the interest. " [to LIDAR]" means sensor_T_LiDAR. In other words, all the sensor measurements are calibrated to the LiDAR sensor on the roof of the vehicle.
>
> Regarding the [...]

Thanks for your reply, i've tried other translations, and quite sure about **sensor_T_LiDAR**.

However i still feel confused with the rotation, Could u please tell me the axis directions of the sensors?
Here is my guess of the sensors' XYZ(along vehicle):
Cam0: RDF (Right-Down-Forward)
Lidar: RFU 
Xsens: FLU
I tried to recover the euler angles from some rotation matrix, such as imu_R_Lidar, Cam0_R_Lidar, the values do not correspond with my guess.

weisongwen, 2020-12-16:
@huanhexiao haha your questions are quite a navigation style. We directly calibrate all the sensors to the LiDAR on the roof and we do not provide coordinate information like "RDF (Right-Down-Forward)". But I  think you can recover this information from the extrinsic. e.g. you can take a look at the LiDAR clouds data. I believe you can figure out this via existing data. 

BTW, @yzhou377  may help but he is pretty busy. Please let us know if you still get problems and show us your **algorithm details.** about how you use the data.

huanhexiao, 2020-12-16:
@weisongwen Thanks for patient and time. 
In addition to my navigation style, i want to estimate the extrinsic paras of IMU-Cam in a recent research, so i need an accurate one as a reference.
 i will make some tests first.
Looking forward @yzhou377  your answer!

Kailthen, 2022-05-17:
I'm ploting point cloud point to cam0(front image), the result  seems not corrent. Any suggestion ?
[image]

weisongwen, 2022-05-30:
Hi @Kailthen , sorry for the delayed reply. You are right, the extrinsic parameters between the camera and the LiDAR are not accurate. During the dataset preparation, the extrinsic parameters are calibrated using the Autoware. You are suggested to use some existing open-sourced work for LiDAR-camera calibration to refine the extrinsic parameters. We will follow this up but It is hard to provide the exact timeline. 

Thank you,
Weisong,

wjf1997, 2023-12-21:
> I am confused with CA data extrinsic paras, or the coordinate axis.
> 
> 1. Could u please tell me the axis directions of each sensor?
> 2. What does each " [to LIDAR]"  mean, **senor_T_Lidar** or **Lidar_T_sensor** ?
> 
> For instance,I guess the axis directions of Lidar seems like Right-Forward-Up(alongVehicle) according to prius.png, and Xsens's is Forward-Left-Up, as Xsens's extr is [0., -1., 0., 0., 1., 0., 0., 0., 0., 0., 1., 7.62e-02] According to pose convention a_T_b, the z-axis translation 7.62e-02 means the b is above a, so **imu_T_Lidar** is right, rather than **Lidar_T_imu**. Based on this supporse, each camera's extr means camera_T_Lidar, and each y-axis translation should be negative value since camera's y-axis defines the vertical direction, right? But the Cam4 and Cam5 is not.
> 
> My discription may seem complicated, but i think i'm talking about some simple issues. So can u do me a favour?

" According to pose convention a_T_b, the z-axis translation 7.62e-02 means the b is above a,"
it should means a is above b, thus 7.62e-02 should be negative, right?