weisongwen/UrbanLoco #11: /navsat/fix and /navsat/odom

eros-ck, 2020-12-17:
Thanks for the great work and sharing with the community.

I think you might need Some specs not clear from the documentation:
- What are `/navsat/fix` and `/navsat/odom` in the rosbags? Are they from NovAtel or ublox?
- What is the difference `/imu_rt` and `/imu_raw`?

weisongwen, 2020-12-27:
Thanks for your interest in our dataset. For the CA dataset, the ```/navsat/fix``` and ```/navsat/odom``` are the middle information which are combined in topic ```/novatel_data/inspvax``` for ground truth evaluation. The ```/imu_rt``` is the imu data from SPAN-CPT but the accuracy is not guaranteed. 

As mentioned on the readme page, the ```/imu_raw``` is the imu data from Xsens IMU.

eros-ck, 2020-12-30:
Thank you @weisongwen. Appreciate it!

closed 2020-12-30 by eros-ck