weisongwen/UrbanLoco #13: where is poor gnss data?

yuxu1997, 2020-12-30:
你好，我下载了你们的香港地区录制的数据集，感谢你们的工作。根据你们的说明，HK-Data20190426-1、HK-Data20190316-2、HK-Data20190316-1都有poor gnss visibilities的特点。但是我从/navsat/fix话题读取的gnss数据却是很好的，如第二幅图所示，而我需要包含野直或者信号不好的地方的gnss数据。所以我想请问是否我弄错了topic，从哪儿可以得到poor gnss data？
[image]
[image]

weisongwen, 2021-01-03:
Hi, Thanks for your interest. Sorry for the delay, we try our best to reply to the issues every several days,

We only provide the raw GNSS data via rostopic "/GNSS_" in the HK dataset. The data you use (/navsat/fix) is the ground truth pose from SPAN-CPT. 
1.  If you want to use raw GNSS data with poor GNSS measurements, you can refer to our team's other dataset here (https://github.com/weisongwen/UrbanNavDataset). 
2. BTW, you can also try the CA dataset directly using the GNSS loose solution via rostopic "/ublox_node/fix".

It can be good if you can present your issue in English which can be other users to track similar issues.

Thanks,

weisongwen, 2021-01-25:
Closed, as no response for this issue.

closed 2021-01-25 by weisongwen

bryantaoli, 2022-04-15:
Thank you Dr.Wen, in Urbanloco website, the HK-Data20190426-1 cantains GNSS (RINEX) (GNSS Raw Data). But it cannot be clicked, could you please update it, thank you agian.

[image]

weisongwen, 2022-04-15:
Solved and closed. Feel free to re-open if necessary,

bryantaoli, 2022-04-15:
> Thank you Dr.Wen, in Urbanloco website, the HK-Data20190426-1 cantains GNSS (RINEX) (GNSS Raw Data). But it cannot be clicked, could you please upload it, thank you agian.
> 
> [image]

Thank @weisongwen to provide the RINEX file of HK-Data20190426-1. 
I upload it below:
[RINEX-HKDate20190426-1.zip](https://github.com/weisongwen/UrbanLoco/files/8496055/RINEX-HKDate20190426-1.zip)