weisongwen/UrbanLoco #15: IMU intrinsics and timestamp of pointcloud

peterWon, 2021-06-04:
Hi, thanks a lot for your great work!

Currently I'm going to find a dataset which contrains LiDAR and IMU to verify my algorithm. That is why I find this wonderful repository. One problem remains is I want to know whether this dataset has timestamp of each point from LiDAR. Beside, It will be appreciated if you could provide the IMU's noise intrinsics.

Sorry for bothering~

weisongwen, 2021-08-11:
Hi @peterWon , Thanks for your interest. The timestamps of the LiDAR point cloud is concerning this frame. Unfortunately, we do not have the timestamps for each point in one frame of point clouds. Regarding the noise of the IMU, please see below:
https://github.com/weisongwen/UrbanNavDataset/blob/master/UrbanNav-HK-Data20200314/xsens_imu_param.yaml

lyaprunov, 2022-03-08:
您好，请问您两个数据集都没有做传感器之间的时间硬同步对吗？

weisongwen, 2022-03-08:
hi @lyaprunov , yes. The hardware synchronization was not done.

lyaprunov, 2022-03-09:
Thank you very much for your reply. I am referring to your paper “Optimization-Based Visual-Inertial SLAM Tightly Coupled with Raw GNSS Measurements”.
Did you not see IMU internal parameter calibration when running CA dataset with Vincs-Mono?  Are the IMU internal parameters of this data set the same as those of HK data set?

weisongwen, 2022-05-30:
Hi @lyaprunov ! sorry for the delayed reply. The dataset from HK and CA share the same IMU model. You can use the one from HK for the CA dataset (although the noise parameters of the IMU may not be highly accurate).  

Thank you,
Weisong,