weisongwen/UrbanLoco #16: how to obtain trajectory from span-cpt

kevin2431, 2021-07-01:
Thanks for sharing the dataset. I want to evaluate my lidar odometry algorithm on this dataset but do not familiar with the GNSS sensor. I wonder if there is a library that can help me convert novatel_msgs/INSPVAX to 6DoF trajectory like the KITTI dataset.

weisongwen, 2021-08-11:
Hi @kevin2431 , Thanks for your interest. You may need some functions to convert the latitude, longitude and altitude to the Eeast, North, Up frames. I have a piece of code that you may need as follows:

https://github.com/weisongwen/GraphGNSSLib/blob/main/global_fusion/include/gnss_tools.h#L220

Best,
Weisong,

Saixiaoma, 2021-12-06:
> Hi @kevin2431 , Thanks for your interest. You may need some functions to convert the latitude, longitude and altitude to the Eeast, North, Up frames. I have a piece of code that you may need as [...]

Hi, how to obtain the rotation(yaw, pitch, roll) from the span-cpt?

weisongwen, 2021-12-06:
you can get the orientation from the topic `/novatel_data/inspvax` which is the rostopic from the span-cpt.

weisongwen, 2022-05-30:
Feel free to re-open if needed.

closed 2022-05-30 by weisongwen

wwtinwhu, 2023-04-02:
Hello, @weisongwen . I used function llh2ecef you provided to convert /novatel_data/bestpos to the xyz. The xy coordinate is similar as the fig which you provided in the README. While, the z values seems not right. I used HK-Data20190117.bag where the GNSS signal is good. Look for help.