weisongwen/UrbanLoco #17: How do I get the ring of robosense

pre-master, 2021-12-29:
Thanks for sharing the dataset . In ca-dataset, your lidar is robosense-32，unlike VLP, the ring of robosense-32 is evenly distributed in the vertical direction according to the fixed resolution. So can you give me some advice on restoring the vertical column information of /rslidar_point in rostopic.(As shown in the figure below, how can I get the arrangement of a ring in the vertical direction)
[image]

Look forward to your reply

pre-master, 2021-12-29:
I solve it

closed 2021-12-29 by pre-master

pytorch1100, 2022-03-09:
Hi@pre-master , I have the same problem. Could you tell me how to solve it?Looking forward to your reply!